pal-robotics/tiago_tutorials #14: Tutorial TIAGo Installation (Indigo)

heitorrapela, 2018-08-03:
Hello, someone could try to install TIAGo Tutorials using Ubuntu 14 + Indigo? My team is facing problems to install it. In the begin of the year, we had installed without problems, but since news update and the delete of master branch, there are some rosinstall link errors and wrong version of .cpp packages, when we try to build it.

ps: I'm without the screenshot at the moment, but if someone couldn't reproduce it, I can re-try and screenshot the error message.

lucamarchionni, 2018-08-03:
We have changed some branches and added as a requirement our version of some ROS packages, but if you use the rosinstall file in the insigo-devel branch of tiago tutorials it will work.

v-lopez, 2019-01-23:
We have updated all our tutorials and are now maintaining only the kinetic version of tiago.

heitorrapela, 2019-01-23:
Ok, so I will close the issue.

closed 2019-01-23 by heitorrapela