pal-robotics/tiago_tutorials #25: Installation Problem during catkin_make

Pranav24-8, 2019-10-17:
Hi, 
Im getting the following errors during catkin_make part of the installation. I get the same errors even if I use catkin build.
```bash
[  2%] Built target aruco_ros_gencfg
Scanning dependencies of target _tf_lookup_generate_messages_check_deps_TfLookupGoal
Scanning dependencies of target _tf_lookup_generate_messages_check_deps_TfStreamActionFeedback
Scanning dependencies of target _tiago_pcl_tutorial_generate_messages_check_deps_PlaneCoefficients
Scanning dependencies of target _tf_lookup_generate_messages_check_deps_lookupTransform
[  2%] Linking CXX shared library /root/tiago_public_ws/devel/lib/libbackward.so
Scanning dependencies of target tiago_pcl_tutorial_gencfg
Scanning dependencies of target _tf_lookup_generate_messages_check_deps_TfLookupActionGoal
In file included from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:34:0,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:34,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:28:
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:58:82: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     CameraParameters(cv::Mat cameraMatrix,cv::Mat distorsionCoeff,cv::Size size) throw(cv::Exception);
                                                                                  ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:64:80: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void setParams(cv::Mat cameraMatrix,cv::Mat distorsionCoeff,cv::Size size) throw(cv::Exception);
                                                                                ^~~~~
In file included from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:34:0,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:34,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:28:
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:79:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void readFromFile(string path)throw(cv::Exception);
                                   ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:82:49: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void saveToFile(string path,bool inXML=true)throw(cv::Exception);
                                                 ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:86:42: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void readFromXMLFile(string filePath)throw(cv::Exception);
                                          ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:90:31: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void resize(cv::Size size)throw(cv::Exception);
                               ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:107:137: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void glGetProjectionMatrix( cv::Size orgImgSize, cv::Size size,double proj_matrix[16],double gnear,double gfar,bool invert=false   )throw(cv::Exception);
                                                                                                                                         ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:119:139: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void OgreGetProjectionMatrix( cv::Size orgImgSize, cv::Size size,double proj_matrix[16],double gnear,double gfar,bool invert=false   )throw(cv::Exception);
                                                                                                                                           ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:125:132: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     static void argConvGLcpara2( double cparam[3][4], int width, int height, double gnear, double gfar, double m[16], bool invert )throw(cv::Exception);
                                                                                                                                    ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/cameraparameters.h:126:96: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     static int  arParamDecompMat( double source[3][4], double cpara[3][4], double trans[3][4] )throw(cv::Exception);
                                                                                                ^~~~~
In file included from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:34:0,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:28:
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:76:104: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void calculateExtrinsics(float markerSize, const CameraParameters &CP, bool setYPerpendicular=true)throw(cv::Exception);
                                                                                                        ^~~~~
In file included from /root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:34:0,
                 from /root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:28:
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:83:129: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void calculateExtrinsics(float markerSize, cv::Mat  CameraMatrix, cv::Mat Distorsion=cv::Mat(), bool setYPerpendicular=true)throw(cv::Exception);
                                                                                                                                 ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:88:61: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void glGetModelViewMatrix(  double modelview_matrix[16])throw(cv::Exception);
                                                             ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/marker.h:100:77: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void OgreGetPoseParameters(  double position[3], double orientation[4] )throw(cv::Exception);
                                                                             ^~~~~
[  2%] Built target _tf_lookup_generate_messages_check_deps_TfStreamActionGoal
In file included from /root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:28:0:
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:89:34: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void saveToFile(string sfile)throw (cv::Exception);
                                  ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/include/aruco/board.h:92:36: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     void readFromFile(string sfile)throw (cv::Exception);
[... 22 lines cut ...]
/root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:67:54: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void BoardConfiguration::saveToFile ( string sfile ) throw ( cv::Exception )
                                                      ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:76:57: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void BoardConfiguration::saveToFile(cv::FileStorage &fs)throw (cv::Exception) {
                                                         ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:97:56: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void BoardConfiguration::readFromFile ( string sfile ) throw ( cv::Exception )
                                                        ^~~~~
/root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:107:59: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void BoardConfiguration::readFromFile(cv::FileStorage &fs)throw (cv::Exception)
[... cut ...]
/root/tiago_public_ws/src/aruco_ros/aruco/src/aruco/board.cpp:210:81: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void Board::OgreGetPoseParameters ( double position[3], double orientation[4] ) throw ( cv::Exception )
                                                                                 ^~~~~
Generating reconfiguration files for Physics in gazebo_ros
Wrote header file in /root/tiago_public_ws/devel/include/gazebo_ros/PhysicsConfig.h
Scanning dependencies of target pcl_msgs_generate_messages_nodejs
```

v-lopez, 2019-10-17:
I don't see any error, those are all warnings. Are you sure you pasted the full log?

Pranav24-8, 2019-10-17:
The full log is too large. I get these warnings constantly and at the end I get the following:
```bash
make[2]: *** [ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/src/joint_trajectory_controller.cpp.o] Error 1
ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/build.make:62: recipe for target 'ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/src/joint_trajectory_controller.cpp.o' failed
CMakeFiles/Makefile2:51404: recipe for target 'ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/all' failed
make[1]: *** [ros_controllers/joint_trajectory_controller/CMakeFiles/joint_trajectory_controller.dir/all] Error 2
make: *** [all] Error 2
Makefile:140: recipe for target 'all' failed
####
#### Running command: "cmake /root/tiago_public_ws/src -DCATKIN_DEVEL_PREFIX=/root/tiago_public_ws/devel -DCMAKE_INSTALL_PREFIX=/root/tiago_public_ws/install -G Unix Makefiles" in "/root/tiago_public_ws/build"
####
####
#### Running command: "make -j40 -l40" in "/root/tiago_public_ws/build"
####
Invoking "make -j40 -l40" failed
```

v-lopez, 2019-10-17:
First make sure you are following all the steps in the tutorials here: http://wiki.ros.org/Robots/TIAGo/Tutorials

Yes, you could attach the whole log here as a file, or paste it on pastebin, but that leads me to believe that you haven't even looked into the issue yourself, since you haven't found the offending line.

We'd appreciate if you took the effort to look in your build log, find the error, and if you are not able to fix it, then report it here so we may help you. But probably it's something that can be fixed on your side.

Pranav24-8, 2019-10-17:
Thank you for the support. I'll look into it and get back to you soon.

Pranav24-8, 2019-10-18:
So I have found the offending lines. The problem seems to be specific to the ros_controllers package. Please have a look at the following.
´´´bash
In file included from /root/catkin_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /root/catkin_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:60:3: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   ResultPtr preallocated_result_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~
   Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:61:3: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   FeedbackPtr preallocated_feedback_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~~~
   Feedback
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:50: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                  ^~~~~~~~~
                                                  Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:115: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                                                                                   ^~~~~~~~~~~
                                                                                                                   Feedback
In file included from /root/catkin_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /root/catkin_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h: In constructor ‘realtime_tools::RealtimeServerGoalHandle<Action>::RealtimeServerGoalHandle(realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle&, const int&, const int&)’:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:68:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_result_’
       preallocated_result_(preallocated_result),
       ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:69:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_feedback_’
       preallocated_feedback_(preallocated_feedback)
       ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: error: ‘preallocated_result_’ was not declared in this scope
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: note: suggested alternative: ‘preallocated_result’
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
          preallocated_result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: error: ‘preallocated_feedback_’ was not declared in this scope
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: note: suggested alternative: ‘preallocated_feedback’
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
          preallocated_feedback
In file included from /root/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/robot_hw.h:35:0,
                 from /root/catkin_ws/src/ros_control/controller_interface/include/controller_interface/controller_base.h:37,
                 from /root/catkin_ws/src/ros_control/controller_interface/include/controller_interface/controller.h:35,
                 from /root/catkin_ws/src/ros_controllers/temperature_sensor_controller/include/temperature_sensor_controller/temperature_sensor_controller.h:4,
                 from /root/catkin_ws/src/ros_controllers/temperature_sensor_controller/src/temperature_sensor_controller.cpp:4:
/root/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h: In instantiation of ‘static char (& hardware_interface::CheckIsResourceManager<T>::callCM(std::vector<C*>&, C*, typename C::resource_manager_type*))[1] [with C = hardware_interface::ActuatorTemperatureSensorInterface; T = hardware_interface::ActuatorTemperatureSensorInterface; hardware_interface::CheckIsResourceManager<T>::yes = char [1]; typename C::resource_manager_type = hardware_interface::ResourceManager<hardware_interface::ActuatorTemperatureSensorHandle>]’:
/root/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:83:14:   required from ‘static const void hardware_interface::CheckIsResourceManager<T>::callConcatManagers(std::vector<T*>&, T*) [with T = hardware_interface::ActuatorTemperatureSensorInterface]’
[... cut ...]
[ 89%] Linking CXX shared library /root/catkin_ws/devel/lib/libgazebo_ros_api_plugin.so
[ 89%] Built target cylinder_detector
[ 89%] Built target gazebo_ros_api_plugin
[ 89%] Built target region_segment
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
####
#### Running command: "cmake /root/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -G Unix Makefiles" in "/root/catkin_ws/build"
####
####
#### Running command: "make -j40 -l40" in "/root/catkin_ws/build"
####
Invoking "make -j40 -l40" failed
´´´

v-lopez, 2019-10-21:
I don't think it has to do with our repos. It looks like it's missing some definitions that were removed here: https://github.com/ros-controls/realtime_tools/commit/6e0e91ef34eb16b3aa95987d35ef65ee7060ea2b#diff-2ee9910bfa291e95139620de52b72eb9L52

Pranav24-8, 2019-10-21:
I'll look into it. Thank you.

Pranav24-8, 2019-10-22:
Looks like your fork of the ros-controllers package is in kinetic-devel. The original one has a melodic-devel which solves this problem. This is what they said when I raised an issue in realtime_tools page.

saikishor, 2019-10-23:
@Pranav24-8 I have tested building the tutorials in an updated melodic docker, and with `catkin_make` I am able to build my packages perfectly. Please find the compilation log here : [https://pastebin.com/Bmm144Uu](https://pastebin.com/Bmm144Uu).

I have these following packages in my workspace:
```
play_motion
aruco_ros	
pmb2_navigation
backward_ros	
pmb2_robot
ddynamic_reconfigure_python
pmb2_simulation
dynamic_introspection
robot_pose
eband_local_planner
roboticsgroup_gazebo_plugins	
ros_control
gazebo_ros_pkgs	
ros_controllers
head_action	
rviz_plugin_covariance
hey5_description
humanoid_msgs
openslam_gmapping
pal_gazebo_plugins
simple_grasping_action
pal_gazebo_worlds
simple_models_robot
pal_gripper	
slam_gmapping
pal_hardware_gazebo
teleop_tools
pal_hardware_interfaces
tf_lookup
pal_msgs	
tiago_description_calibration
pal_navigation_cfg_public 
tiago_moveit_config
pal_navigation_sm
tiago_navigation
pal_python	
tiago_robot
pal_statistics	
tiago_simulation
pal_wsg_gripper	
tiago_tutorials
```
I have `ros-melodic-realtime-tools(1.15.0-1bionic.20190809.155508)` installed. Could you please clean your workspace and try rebuilding.

Pranav24-8, 2019-10-23:
Alright thank you. I will try and let you know about the results.

saikishor, 2019-10-23:
@Pranav24-8 Sure. Keep us posted.

Pranav24-8, 2019-10-23:
@saikishor can you also please post your Dockerfile here? I am also having this problem only while I am building it in a docker image.

saikishor, 2019-10-23:
@Pranav24-8 I used the official docker for testing `osrf/ros:melodic-desktop-full`. Please try using this.

Pranav24-8, 2019-10-23:
Alright thank you. Will try this.

Pranav24-8, 2019-10-23:
Installing an updated version of ros-melodic-desktop-full did the trick. Thank you for the help guys.

closed 2019-10-23 by Pranav24-8