pal-robotics/tiago_tutorials #27: Problems running  Tiago Simulation from tutorial

EspenTeigen, 2020-10-05:
Hi.

I have followed the [tutorial on ROS]( http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/TiagoSimulation) and have some problems running the examples.

Attached is the output from the terminal when i run 
'roslaunch tiago_gazebo tiago_gazebo.launch public_sim:=true robot:=steel'

[tiago_gazebo.launch.LOG](https://github.com/pal-robotics/tiago_tutorials/files/5326110/tiago_gazebo.launch.LOG)

Here is the output from rqt_graph

[image]

And here is the project tree(I have made no changes to this after downloading)

[project_tree.LOG](https://github.com/pal-robotics/tiago_tutorials/files/5326140/project_tree.LOG)

v-lopez, 2020-10-05:
Can you make sure all the packages in the workspace are compiled? In the project tree I don't see the build directories for `play_motion`.

EspenTeigen, 2020-10-05:
After running catkin make again, I got some error messages i did not notice earlier. Here are the log

[build.cmake.002.log](https://github.com/pal-robotics/tiago_tutorials/files/5326294/build.cmake.002.log)

v-lopez, 2020-10-05:
It looks like some errors finding /usr/lib/x86_64-linux-gnu/libEGL.so 

See if you can fix it and if you have any issues after that please let us know.