pal-robotics/tiago_tutorials #28: Installation problem

Cmeo97, 2021-03-17:
Hi! I followed installation tutorial, I have Ubuntu 18.04.5 LTS, I built it without problem or warnings, it says: 
[build] Summary: All 150 packages succeeded!                                   
[build]   Ignored:   18 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.    

but when I try to launch the simulation I cannot see the robot and I got some errors:

 roslaunch tiago_dual_gazebo tiago_dual_gazebo.launch public_sim:=true end_effector_left:=pal-gripper end_effector_right:=pal-gripper
... logging to /home/cristian/.ros/log/05e97d98-874f-11eb-9bb1-e4e74942086e/roslaunch-cristian-OMEN-by-HP-Laptop-15-dc0xxx-15247.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://cristian-OMEN-by-HP-Laptop-15-dc0xxx:36145/

SUMMARY
========

PARAMETERS
 * /arm_left_controller/constraints/arm_left_1_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_2_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_3_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_4_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_5_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_6_joint/goal: 0.02
 * /arm_left_controller/constraints/arm_left_7_joint/goal: 0.02
 * /arm_left_controller/constraints/goal_time: 0.6
 * /arm_left_controller/constraints/stopped_velocity_tolerance: 5.0
 * /arm_left_controller/joints: ['arm_left_1_join...
 * /arm_left_controller/type: position_controll...
 * /arm_right_controller/constraints/arm_right_1_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_2_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_3_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_4_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_5_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_6_joint/goal: 0.02
 * /arm_right_controller/constraints/arm_right_7_joint/goal: 0.02
 * /arm_right_controller/constraints/goal_time: 0.6
 * /arm_right_controller/constraints/stopped_velocity_tolerance: 5.0
 * /arm_right_controller/joints: ['arm_right_1_joi...
 * /arm_right_controller/type: position_controll...
 * /collision_model_parameters/blacklist/default: [{'second': 'tors...
 * /collision_model_parameters/links_with_padding: []
 * /collision_model_parameters/padding: 0
 * /force_torque/wrist_left_ft/frame: wrist_left_ft_link
 * /force_torque/wrist_left_ft/sensor_joint: arm_left_7_joint
 * /force_torque/wrist_right_ft/frame: wrist_right_ft_link
 * /force_torque/wrist_right_ft/sensor_joint: arm_right_7_joint
 * /force_torque_sensor_controller/publish_rate: 50
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /gains/FFJ1/d: 0
 * /gains/FFJ1/i: 10
 * /gains/FFJ1/p: 100
 * /gains/FFJ2/d: 0
 * /gains/FFJ2/i: 10
 * /gains/FFJ2/p: 100
 * /gains/FFJ3/d: 0
 * /gains/FFJ3/i: 10
 * /gains/FFJ3/p: 100
 * /gains/FFJ4/d: 0
 * /gains/FFJ4/i: 10
 * /gains/FFJ4/p: 100
 * /gains/MFJ1/d: 0
 * /gains/MFJ1/i: 10
 * /gains/MFJ1/p: 100
 * /gains/MFJ2/d: 0
 * /gains/MFJ2/i: 10
[... 865 lines cut ...]
 * /robot_description_planning/joint_limits/gripper_right_left_finger_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/gripper_right_left_finger_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/gripper_right_left_finger_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/gripper_right_left_finger_joint/max_velocity: 0.05
 * /robot_description_planning/joint_limits/gripper_right_right_finger_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/gripper_right_right_finger_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/gripper_right_right_finger_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/gripper_right_right_finger_joint/max_velocity: 0.05
 * /robot_description_planning/joint_limits/hand_right_index_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/hand_right_index_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/hand_right_index_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/hand_right_index_joint/max_velocity: 3.35
 * /robot_description_planning/joint_limits/hand_right_mrl_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/hand_right_mrl_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/hand_right_mrl_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/hand_right_mrl_joint/max_velocity: 3.56
 * /robot_description_planning/joint_limits/hand_right_thumb_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/hand_right_thumb_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/hand_right_thumb_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/hand_right_thumb_joint/max_velocity: 3.35
 * /robot_description_planning/joint_limits/torso_lift_joint/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/torso_lift_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/torso_lift_joint/max_acceleration: 0
 * /robot_description_planning/joint_limits/torso_lift_joint/max_velocity: 0.07
 * /robot_description_semantic:

saikishor, 2021-03-18:
Hello @Cmeo97,
Thanks for writing to us. From the log that you have attached, it seems the issue is arising due to the Python3 setup in your machine. 
You are facing the issue because tf2_ros was compiled for Python2 environment similar to the default ROS Melodic setup. You can find a detailed information here: https://github.com/ros/geometry2/issues/259

saikishor, 2021-04-26:
@Cmeo97 I will be closing the issue due to inactivity for more than 30 days. We hope you fixed your issue and having great time with our TIAGo simulation.

Thank you

closed 2021-04-26 by saikishor