pal-robotics/tiago_tutorials #29: Installations problems

B-Acharya, 2021-04-24:
I followed the tutorial https://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/TiagoSimulation for the installations but the compilation is crashing 
[build] Summary: 8 of 139 packages succeeded.                                                                                                       
[build]   Ignored:   15 packages were skipped or are blacklisted.                                                                                   
[build]   Warnings:  10 packages succeeded with warnings.                                                                                           
[build]   Abandoned: 129 packages were abandoned.                                                                                                   
[build]   Failed:    2 packages failed.                                                                                                             
[build] Runtime: 2.3 seconds total.   

i am currently running Ubuntu 18.04 and ROS melodic

saikishor, 2021-04-24:
@B-Acharya Thanks for showing interest in our TIAGo Tutorials. 

Could you please give us more information on the package you are having issue compiling?. Have you followed all the steps mentioned in the page?.

Thank you,

B-Acharya, 2021-04-24:
@saikishor  i was following the step by step installation guidelines provided in the tutorials page, after i run
` rosdep install --from-paths src --ignore-src -y --rosdistro melodic --skip-keys="opencv2 opencv2-nonfree pal_laser_filters speed_limit_node sensor_to_cloud hokuyo_node libdw-dev python-graphitesend-pip python-statsd pal_filters pal_vo_server pal_usb_utils pal_pcl pal_pcl_points_throttle_and_filter pal_karto pal_local_joint_control camera_calibration_files pal_startup_msgs pal-orbbec-openni2 dummy_actuators_manager pal_local_planner gravity_compensation_controller current_limit_controller dynamic_footprint dynamixel_cpp tf_lookup opencv3"`
i get the output saying all dependensces have been installed and then when i try to run 
`catkin build -DCATKIN_ENABLE_TESTING=0`
i get the error that i mentioned above . do you need the complete output of the build process ?

Thanks for the quick reply

B-Acharya, 2021-04-24:
Now i tried to repeat the process and i get a different error when i run 
`rosinstall src /opt/ros/melodic tiago_dual_public-melodic.rosinstall`

ERROR in config: Invalid multiproject yaml format in [tiago_dual_public-melodic.rosinstall]: mapping values are not allowed here
  in "tiago_dual_public-melodic.rosinstall", line 98, column 65

saikishor, 2021-04-24:
@B-Acharya Thanks for getting back to us. 

Could you share the output of the `rosdep install` command and also the the compilation error log(first error should be sufficient)?.

Have you done `sudo apt-get update` before running the rosdep install?. If not, could you try doing the same?.

Thank you,

B-Acharya, 2021-04-24:
`rosdep install --from-paths src --ignore-src -y --rosdistro melodic --skip-keys="opencv2 opencv2-nonfree pal_laser_filters speed_limit_node sensor_to_cloud hokuyo_node libdw-dev python-graphitesend-pip python-statsd pal_filters pal_vo_server pal_usb_utils pal_pcl pal_pcl_points_throttle_and_filter pal_karto pal_local_joint_control camera_calibration_files pal_startup_msgs pal-orbbec-openni2 dummy_actuators_manager pal_local_planner gravity_compensation_controller current_limit_controller dynamic_footprint dynamixel_cpp  opencv3"`
#All required rosdeps installed successfully
This is the error i get when i run the catkin build 
fatal error: opencv2/core/core.hpp: No such file or director

But this is  confusing as shouldn't rosdep install all the dependencies ?

saikishor, 2021-04-24:
@B-Acharya Yes you are right rosdep should install all dependencies of the packages, seems that they are properly installed in your machine, however, it seems you have some issue with your opencv libraries, that's the reason the compilation is failing.

saikishor, 2021-04-28:
@B-Acharya Were you able to figure out the issue?. If not, feel free to get back to us with more info.

Thank you,

saikishor, 2021-04-28:
@B-Acharya We have recently added the option to try our TIAGo simulation using dockers. This is pretty easy to setup and you will be able to launch the simulation and start working with it in minutes.

Please kindly visit the section **"Docker-based installation"** in the page : [http://wiki.ros.org/Robots/TIAGo/melodic_install](http://wiki.ros.org/Robots/TIAGo/melodic_install).

We hope this helps you to solve your issue. 

Thank you,

saikishor, 2021-05-06:
@B-Acharya We hope you have fixed the issue and have been using our TIAGo Tutorials.

We wish you a great joyful time with our TIAGo simulation!

Best Regards,
PAL Robotics Team.

closed 2021-05-06 by saikishor