pal-robotics/tiago_tutorials #30: Installation Error

shrirangsp, 2021-05-05:
Hello,

Configuration: Ubuntu 16.04 + ROS Kinetic

`Errors     << transmission_interface:make /home/ros/tiago_public_ws/logs/transmission_interface/build.make.001.log                                                                                         
In file included from /home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:34:0,
                 from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/include/transmission_interface/transmission_loader.h:48,
                 from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/src/transmission_loader.cpp:32:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:54:71: error: ‘nullptr’ was not declared in this scope
                       const double* eff, const double* absolute_pos = nullptr,
                                                                       ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:55:53: error: ‘nullptr’ was not declared in this scope
                       const double* torque_sensor = nullptr,
                                                     ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:56:62: error: ‘nullptr’ was not declared in this scope
                       const std::vector<double>* pid_gains = nullptr,
                                                              ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:57:47: error: ‘nullptr’ was not declared in this scope
                       const double* ff_term = nullptr)
                                               ^
In file included from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/include/transmission_interface/transmission_loader.h:48:0,
                 from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/src/transmission_loader.cpp:32:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:55:55: error: ‘nullptr’ was not declared in this scope
                  std::vector<double>* pid_gains_cmd = nullptr, double* ff_term_cmd = nullptr)
                                                       ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:55:86: error: ‘nullptr’ was not declared in this scope
                  std::vector<double>* pid_gains_cmd = nullptr, double* ff_term_cmd = nullptr)
                                                                                      ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h: In constructor ‘hardware_interface::ActuatorHandle::ActuatorHandle()’:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:44:35: error: ‘nullptr’ was not declared in this scope
     : ActuatorStateHandle(), cmd_(nullptr), pid_gains_cmd_(nullptr), ff_term_cmd_(nullptr)
                                   ^
make[2]: *** [CMakeFiles/transmission_interface_loader.dir/src/transmission_loader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/include/transmission_interface/transmission_interface_loader.h:52:0,
                 from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/src/transmission_interface_loader.cpp:33:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:54:71: error: ‘nullptr’ was not declared in this scope
                       const double* eff, const double* absolute_pos = nullptr,
                                                                       ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:55:53: error: ‘nullptr’ was not declared in this scope
                       const double* torque_sensor = nullptr,
                                                     ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:56:62: error: ‘nullptr’ was not declared in this scope
                       const std::vector<double>* pid_gains = nullptr,
                                                              ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_state_interface.h:57:47: error: ‘nullptr’ was not declared in this scope
                       const double* ff_term = nullptr)
                                               ^
In file included from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/include/transmission_interface/transmission_interface_loader.h:53:0,
                 from /home/ros/tiago_public_ws/src/ros_control/transmission_interface/src/transmission_interface_loader.cpp:33:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:55:55: error: ‘nullptr’ was not declared in this scope
                  std::vector<double>* pid_gains_cmd = nullptr, double* ff_term_cmd = nullptr)
                                                       ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:55:86: error: ‘nullptr’ was not declared in this scope
                  std::vector<double>* pid_gains_cmd = nullptr, double* ff_term_cmd = nullptr)
                                                                                      ^
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h: In constructor ‘hardware_interface::ActuatorHandle::ActuatorHandle()’:
/home/ros/tiago_public_ws/src/ros_control/hardware_interface/include/hardware_interface/actuator_command_interface.h:44:35: error: ‘nullptr’ was not declared in this scope
     : ActuatorStateHandle(), cmd_(nullptr), pid_gains_cmd_(nullptr), ff_term_cmd_(nullptr)
                                   ^
make[2]: *** [CMakeFiles/transmission_interface_loader.dir/src/transmission_interface_loader.cpp.o] Error 1
make[1]: *** [CMakeFiles/transmission_interface_loader.dir/all] Error 2
make: *** [all] Error 2
cd /home/ros/tiago_public_ws/build/transmission_interface; catkin build --get-env transmission_interface | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << transmission_interface:make                                    [ Exited with code 2 ]                                                                                                        
Failed    <<< transmission_interface                                         [ 2.3 seconds ]                                                                                                               
Abandoned <<< pal_carbon_collector                                           [ Unrelated job failed ]                                                                                                      
Abandoned <<< pal_statistics                                                 [ Unrelated job failed ]                                                                                                      
Abandoned <<< simple_models_description                                      [ Unrelated job failed ]                                                                                                      
[... 49 lines cut ...]
Abandoned <<< tiago_controller_configuration                                 [ Unrelated job failed ]                                                                                                      
Abandoned <<< tiago_controller_configuration_gazebo                          [ Unrelated job failed ]                                                                                                      
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 1 minute and 20.6 seconds total.                                                                                                                                                          
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`

Kindly help in resolving this errors.

Thank you,
Shrirang

saikishor, 2021-05-05:
@shrirangsp Thanks for showing interest in our TIAGo tutorials. It seems one of our recent change has broken the Kinetic tutorials, we will be fixing it soon by this week. Thank you for pointing it out. Sorry for the inconvenience caused. 

We have recently added the option to try our TIAGo simulation using dockers. This is pretty easy to setup and you will be able to launch the simulation and start working with it in minutes.

Please kindly visit the section "Docker-based installation" in the page : http://wiki.ros.org/Robots/TIAGo/melodic_install.

We hope this helps you to continue using our TIAGo tutorials.

Thank you,

saikishor, 2021-05-10:
@shrirangsp The error you have mentioned in this issue, have been fixed with the recent changes pushed to our GitHub on last Friday. Could you please pull the recent fixes and give it a try?.

Best Regards,
PAL Robotics Team.

saikishor, 2021-05-12:
Hello @shrirangsp ,

We hope your issue is resolved with the recent changes pushed to our GitHub repositories. We will be closing this issue on this regard. Have a fruitful time with our TIAGo Simulation. 

Best Regards,
PAL Robotics Team.

closed 2021-05-12 by saikishor