pal-robotics/tiago_tutorials #33: Tiago docker : problem with Rocker DeprecationWarning: distro.linux_distribution

DavidNguyen95, 2021-08-01:
I try to pull tiago docker images and install rocker. follow 3 steps:(step1,2 are work but error in step 3)
1. `docker pull palroboticssl/tiago_tutorials:melodic
`
2. `distribution=$(. /etc/os-release;echo $ID$VERSION_ID)    && curl -s -L https://nvidia.github.io/nvidia-docker/gpgkey | sudo apt-key add -    && curl -s -L https://nvidia.github.io/nvidia-docker/$distribution/nvidia-docker.list | sudo tee /etc/apt/sources.list.d/nvidia-docker.list
sudo apt-get update
sudo apt-get install -y nvidia-docker2
sudo systemctl restart docker`
3. `rocker --nvidia --x11 palroboticssl/tiago_tutorials:melodic
`
but in third step i got error 
`File "/home/francesco/.local/bin/rocker", line 8, in <module>
    sys.exit(main())
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/cli.py", line 65, in main
    dig = DockerImageGenerator(active_extensions, args_dict, base_image)
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/core.py", line 202, in __init__
    self.dockerfile = generate_dockerfile(active_extensions, self.cliargs, base_image)
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/core.py", line 319, in generate_dockerfile
    dockerfile_str += el.get_preamble(args_dict) + '\n'
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/nvidia_extension.py", line 117, in get_preamble
    return em.expand(preamble, self.get_environment_subs(cliargs))
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/nvidia_extension.py", line 97, in get_environment_subs
    detected_os = detect_os(cliargs['base_image'], print, nocache=cliargs.get('nocache', False))
  File "/home/francesco/.local/lib/python3.8/site-packages/rocker/os_detector.py", line 70, in detect_os
    _detect_os_cache[image_name] = literal_eval(output.strip())
  File "/usr/lib/python3.8/ast.py", line 59, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.8/ast.py", line 47, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 1
    detect_os.py:1: DeprecationWarning: distro.linux_distribution() is deprecated. It should only be used as a compatibility shim with Python's platform.linux_distribution(). Please use distro.id(), distro.version() and distro.name() instead.
                ^
SyntaxError: invalid syntax
`
I also have one more question: my pc have GPU (GeForce RTX 208o) so how can use GPU (cuda) inside tiago docker?
Thank you

saikishor, 2021-08-02:
Hello @DavidNguyen95 ,

Thanks for showing interest in our TIAGo Simulation. The above issue seems to be something from the setup of the rocker side, but not from the docker we have provided.

Regarding your question of how to use CUDA inside the TIAGo docker. The TIAGo docker is not built with CUDA libaries, so you might not be able to use CUDA related stuff inside the docker right away. However, if that's your interest, you can write your own Dockerfile, and use our TIAGo docker as a base image and install the CUDA libraries and then use the built image to start using CUDA capabilities inside the docker along with our TIAGo Simulation.

Best Regards,
PAL Robotics Team.

DavidNguyen95, 2021-08-06:
Thank you! After fix rocker the docker container works fine

closed 2021-08-06 by DavidNguyen95