pal-robotics/tiago_tutorials #38: error running ros2 launch tiago_gazebo tiago_gazebo.launch.py

AdrianCobo, 2022-10-06:
# Type of bug
- [ ] A) Typo/error in Figure/text
- [X] B) Bug in code
- [ ] C) General comment, recommendation or petition 

# Issue Description
ros2 launch tiago_gazebo tiago_gazebo.launch.py not working

# A) *Typo/Error Description*
Waiting for '/controller_manager' node to exist
joy_teleop not working

## Screenshot
[image]
[image]
[image]
[image]

## how to reproduce
Steps to reproduce the behavior:
1. run in a terminal: ros2 launch br2_tiago sim.launch.py

## Desktop :
 - OS: [Ubuntu 22.04]
 - ROS2 Distro [Humble]

AdrianCobo, 2022-10-06:
okay, thanks. 

How i have to use the [tiago_public.rosinstall](https://github.com/pal-robotics/tiago_tutorials/blob/humble-devel/tiago_public.rosinstall) ? Sorry for the question but i have never used it before.

Have a nice day,
Adrián

saikishor, 2022-10-06:
Hello @AdrianCobo,

You can refer to this page for more info: http://wiki.ros.org/vcstool.

We hope this helps. We are looking forward for you to use our TIAGo simulation in ROS2.

Best Regards,
PAL Robotics Team.

AdrianCobo, 2022-10-06:
thanks, 
but now I am getting this error:
1665090451.7330711 [INFO] [launch]: All log files can be found below /home/adrian/.ros/log/2022-10-06-23-07-31-732528-PORTATIL-18169
1665090451.7332942 [INFO] [launch]: Default logging verbosity is set to INFO
1665090451.9130342 [ERROR] [launch]: Caught exception in launch (see debug for traceback): No module named 'launch_param_builder'

Thanks for all your help,
Adrian

saikishor, 2022-10-07:
Hello @AdrianCobo,

You are missing to install the package [launch_param_builder](https://github.com/PickNikRobotics/launch_param_builder). Please try to install the package or install all dependencies using rosdep. We hope this helps to solve your issue.

Thank you,

Best Regards,
PAL Robotics Team.

AdrianCobo, 2022-10-07:
All done but now i am getting the same initial issue

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Noel215, 2022-10-07:
Hi @AdrianCobo, 

Execute `dpkg -l | grep humble | grep control`. I think you could be missing `ros-humble-gazebo-ros2-control` package.
If not, try to clean your workspace and be sure that you are using Cyclone DDS.

For the next time, it's better for us that you paste the log instead of screenshots because it's easier to parse.

AdrianCobo, 2022-10-07:
oh sorry,
yeh you were right, that package was missing but I installed it and now the simulation is correctly working!!!

But  now i have other problem; the robot does not move using:
-ros2 launch tiago_gazebo tiago_gazebo.launch.py world_name:=factory
                                     or
-ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -r cmd_vel:=key_vel

:( 

Thank you so much for all the help!!!

saikishor, 2022-10-07:
Hello @AdrianCobo,

We are glad that it worked. Regarding the velocity, try to publish the velocity using the command line and see if you can publish. The topic to publish is `/mobile_base_controller/cmd_vel_unstamped`. We hope this helps.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-10-07 by saikishor

AdrianCobo, 2022-10-07:
yea, i can publish but it dosnt move.

Now i see that i am getting this error:

[gzserver-1] [ERROR] [1665143870.690903681] [controller_manager]: The 'type' param was not defined for 'torso_controller'.
[ERROR] [spawner-7]: process has died [pid 4697, exit code 1, cmd '/home/adrian/Escritorio/tiagosim_ws/install/controller_manager/lib/controller_manager/spawner torso_controller --controller-manager controller_manager --controller-type joint_trajectory_controller/JointTrajectoryController --param-file /home/adrian/Escritorio/tiagosim_ws/install/tiago_controller_configuration/share/tiago_controller_configuration/config/torso_controller.yaml --ros-args'].

AdrianCobo, 2022-10-09:
Yeh it works using ros2 topic pub -r 10 /mobile_base_controller/cmd_vel_unstamped geometry_msgs/Twist "linear: {x: 0.3}"

Thans for all!!!

saikishor, 2022-10-09:
Hello @AdrianCobo,

We are glad you could work it out.

Best Regards,
PAL Robotics Team.