pal-robotics/tiago_tutorials #40: noetic rosinstall error with teleop_tools

gerardcanal, 2023-02-14:
Hi there,
Just realised the opensource simulations for tiago were released for noetic and gave it a go. I found that teleop_tools is set to get the noetic-devel branch in https://github.com/pal-robotics/tiago_tutorials/blob/18bbae1f29eee2ebe81c12e8449f602a1632ef75/tiago_public-noetic.rosinstall#L39 but the teleop_tools github doesn't seem to have the noetic_devel branch (or it has not yet been pushed).

It doesn't seem to be a big issue as it defaults to the default branch (and at least key_teleop seems to work).

saikishor, 2023-02-15:
Hello @gerardcanal,

Thank you for pointing it out. We are already working on the possible fixes and update to our .rosinstall. You can expect an update in the upcoming days.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2023-02-16:
Hello @gerardcanal,

The issue is currently resolved with the new changes. Thank you for your patience. 

Best Regards,
PAL Robotics Team.

closed 2023-02-16 by saikishor