pal-robotics/tiago_tutorials #41: Controller arm_controller aborted.(ros-melodic tiago_pick_demo tutorial script execution)

yeshwanthguru, 2023-03-15:
[image]
when I try to execute the tiago pick demo tutorial this happens, Any clue regarding this . But I haven't faced this before . But the tuck am script works well in the other worlds, when it comes to the tabletop cube world it faces the error even though all parameters are ok.

Thank you

saikishor, 2023-03-15:
Hello @yeshwanthguru,

Thank you for showing interest in our TIAGo simulation. We have tried to reproduce your issue but unable to do so, we would need more information to reproduce it on our end. We also recommend you to take a look at our [Public Simulation Dockers](http://wiki.ros.org/Robots/TIAGo/Tutorials/Installation/Installing_Tiago_tutorial_docker) for your use.

Thank you,

Best Regards,
PAL Robotics Team.

yeshwanthguru, 2023-03-16:
Greetings Team,

I am writing to inform you that I have attached a reference below regarding
an error I encountered while working with the tiago_pick_demo package.
Although it is not a significant issue, I may need to adjust some
parameters to resolve it. I am uncertain whether the problem stems from my
end or not.

When I created the tiago workspace and began working with the
tiago_pick_demo(aruco-based pick and place task) package, I encountered
this problem. However, I did not face any similar issues when executing the
script in other packages. While I could create a separate map and complete
the task, I want to ensure that the issue is not on my end since I created
the package for the first time.I hope you understood my concern, if not let
me know.

Best regards,
Yeshwanth Guru Krishnakumar
 hi-2023-03-16_09.08.48.mp4
<https://drive.google.com/file/d/11_rM4zHzZGL8odfbvS3WzojSOo5jY7Lm/view?usp=drive_web>
 hi-2023-03-16_09.10.45.mp4
<https://drive.google.com/file/d/1F1ZimUE7IGABoEAUDYHFVXDWmVQnXh9C/view?usp=drive_web>

saikishor, 2023-03-27:
Hello @yeshwanthguru,

We apologize for the delay in the response. Yes, we could reproduce it after several tries. We are trying to fix it in a couple of other worlds and other more parametric things. You can edit the world to spawn the table further from the robot. We will try to fix and publish the worlds in the next couple of weeks. 

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2023-03-27:
Hello @yeshwanthguru,

We are glad that you have resolved the issue on your end. We will fix it on our end in a couple of weeks. 

Thank you,

Best Regards,
PAL Robotics Team.